efabless/caravel #521: close missing DRC rules in klayout deck

jeffdi, 2024-01-10:


jeffdi, 2024-01-10:
issue surfaced on 2311 projects for hvi and nwell

jeffdi, 2024-01-10:
received violator cells from SWT, but they don't cover all the rules and we are missing Calibre reports to benchmark results.  The reported are numerous.

RTimothyEdwards, 2024-01-10:
We can implement rules for ourselves based on the s8 calibre rules and the derived (generated) layer tables.  But the generated layer tables are not part of the public documentation, so they would be for internal use only.

However, there are various rules in the public documentation that affect the generated layers.  In some cases, these rules effectively _become_ MR rules in the absence of any direct check of the mask layer itself.  The rule nwell.9 (when HVI intersects nwell, the nwell must be completely covered by HVI) is one such rule.   We need to add that rule to our own MR deck because we don't have MR rules for the generated layers.  There are probably other similar rules that cover other generated layers.

marwaneltoukhy, 2024-04-14:
> The current ruleset incorrectly checks MR_capm.SP.2 rule. The attached PDF shows how the rule should be applied.
> 
> [capm_rules.pdf](https://github.com/efabless/caravel/files/14949772/capm_rules.pdf)

@DavidRLindley there is still no explanation of what is met3_bottom_plate

kareefardi, 2024-04-14:
@DavidRLindley Couple of follow up questions:
- Just to confirm `Metal3 not continuous under CAPM, but electrically connected` implies connectivity through other metal layers e.g.: met3 -> via2 -> met2 -> via2 -> met3
- In scenario 3, does `MR_capm.SP.2` apply to spacing between `Met3_bot_plate` and a `met3` patch that doesn't enclose `capm`? There is such a case in the violator cells.

DavidRLindley, 2024-04-15:
Hi Marwan,
All the Skywater documentation is pretty clear that met3_bottom_plate is
Met3 AND capm grown by the minimum enclosure of capm by Met3 (0.14um).

Hi Kareem,
I think the answer is yes to both questions.  Your first question is a very
good one.  I should not have limited my scenario by Met3 only connection.

I will verify my answers with Skywater.

David

DavidRLindley, 2024-04-15:
Marwan, Kareem,

I verified the following with Skywater.
1. met3_bot_plate is defined as (capm:dg AND met3:dg) sized by 0.14um
2. the capm spacing (MR_capm.SP.1) needs to be satisfied regardless of how the metal3 geoms are connected.
3. MR_capm.SP.2 applies to any met3 that is not connected to the met3_bot_plate.

David

DavidRLindley, 2024-04-16:
> Thanks David. Should we assume that similar rules apply for cap2m?

Yes.

DavidRLindley, 2024-04-22:
Two psdm rules are being flagged on some proprietary memory macro IP.  

1. psdm.4  : min. spacing across areaid:ce : 0.38um
    I think this rule is not applied correctly since both psdm geoms are inside areaid:ce.

2. psdm.6a : min. psdm spacing on parallel edges in areaid:ce : 0.38um
    FAB looked at the attached PDF and agreed that our psdm.6a rule did not apply because the edges were not parallel.  The correct 
    rule to apply in this scenario is the lesser core rule 0.29um.  

We can have a Zoom meeting if you want to view the errors. 

[psdm_rules.pdf](https://github.com/efabless/caravel/files/15067315/psdm_rules.pdf)

DavidRLindley, 2024-05-08:
As of 2024/May/08 these issues exist in the rules
1. mcon straddling the areaid:core is truncated by the rules
2. I have a couple examples in Chipalooza_1 where the a couple of licon shapes are being incorrectly flagged
3. We need an global li density check.
4. We need to settle on the final set of capm rules. 
5. The switches need to be address.  They didn't work correctly in my last run of the most recent rules (core PSDM/NSDM).

jeffdi, 2024-05-22:
need to include new issues related to created layer DRC errors.

jeffdi, 2024-05-22:
remaining issues

- licon
- mcon
- created layers rules - need to generate from reversed engineer work by Tim as well as violator

verify if new rule deck has created layer rules - then check for violator cells

DavidRLindley, 2024-09-25:
All current MR rules have been added.   Further work needed but new tickets to be opened.

closed 2024-09-25 by DavidRLindley