efabless/caravel #524: 3.0 Voltage for vccd fails on silicon

marwaneltoukhy, 2024-01-18:
3.0v for vccd fails all tests on silicon, however 3.1v passes.

RTimothyEdwards, 2024-01-18:
@marwaneltoukhy :  Can you be any more specific about the mechanism of failure?

RTimothyEdwards, 2024-01-18:
Never mind. . .  This is very easy to reproduce.  My chip under test fails intermittently at 3.00V but reliably at 2.95V.

RTimothyEdwards, 2024-01-18:
Problem solved.  It is caused by the component U4 on the development board that is driving the RESETB pin and has a trip point of 2.9V.  This chip should never have been put on the board in the first place (and I said so during board review).  It seems to have been added to the board due to a complete misunderstanding of the reset mechanism of the chip.