efabless/caravel #558: Solve poly density issues

DavidRLindley, 2024-12-05:


DavidRLindley, 2024-12-05:
I have created new fill cells for the sky130_fd_sc_hd library.  I copied the sky130_ef_sc_hd__decap_12 cell and created cells with 80%, 60%, 40%, and 20% poly density of the original cell (the original cell is normalized to 100%).  Along with the existing newfill_12 cell, we're covered from 0% - 100% density. I also created a sky130_ef_sc_hd__fill_2 cell.   We now have sky130_ef_sc_hd__fill_8, sky130_ef_sc_hd__fill_4, sky130_ef_sc_hd__fill_2 cells which match the width of the sky130_fd_sc_hd__fill_8, _4, and _2 cells.  The difference being the ef fill cells have diffusion in them.  We can leave the sky130_fd_sc_hd__fill.... cells in the library but recommend our customers avoid them due to likely FOM density issues.  I don't think the fill algorithm will place FOM fill shapes in the fd fill cells because the nwell height and space are too short, and maybe not place P1M fill shapes either.

There is a sky130_ef_sc_hd__fill_12 cell in the library that I recommend we also recommend our customers avoid.  It causes nwell spacing issues when PNRed.

You can see the different cells in
https://docs.google.com/spreadsheets/d/1vu8994bgB_ctbLMw38fe3nuL451c2kPnJJg7r_ZKwOc/edit?gid=0#gid=0

I ran experiments with the different cells on an 11 sample set of the 2411 chips.  These included a wide spread of decap_12 cell counts (290k - +500k).  All the chips in the test passed poly density with approximately 6.5% margin.  These results are also in the above Sheet link.

Three things need to happen to complete this ticket.
1. Install the new cells in the sky130 PDKs  : Tim Edwards
2. Modify the openlane configuration file to prioritize the sky130_ef_sc_hd__decap_40_12 cell for decap fill.  : Anton
3. The flow needs to be tested to verify the new cells do not introduce unforeseen DRCs, and they consistently solve the poly density problem.  : Anton or David Lindley

RTimothyEdwards, 2024-12-06:
@DavidRLindley :  Please send me the GDS for these cells (and any other views, if you happen to have any, although I can work with just the GDS), and I will incorporate them into open_pdks.

The `sky130_ef_sc_hd__fill_12` could be modified to avoid the nwell error, but it is actually a decap cell and should not be listed as a fill cell.  I could generate a new `fill_12` cell that is, effectively, a `fill_8` abutted to a `fill_4` (that is, not a decap).

I have a `newfill_12` cell as well, which is one we used previously to work around density issues.  Is that now deprecated?

DavidRLindley, 2024-12-06:
Sent via Slack.

DavidRLindley, 2024-12-09:
Sorry, @RTimothyEdwards, I didn't fully read your message.  

The sky130_ef_sc_hd__newfill_12 cell should be replace the sky130_ef_sc_hd__fill_12 cell.

DavidRLindley, 2024-12-18:
Tim has revised open_pdk and it has been pushed into volare.  Next is to write a knowledge base article describing the cells and flow necessary to fix poly density issues when they arise.